bitcraze/discussions #662: Stabalizing a single drone to a defined height using python´s API

AngelS18, 2023-04-14:
I was wondering how I can stabilize a crazyflie to a defined height (by manually inputting the pitch, roll, yaw, and thrust) using Python´s API, but without the implementation of a positioning system such as LOCO or the optic flow deck. The documentation states this can be achieved by using this class method "crazyflie.commander.send_setpoint(roll, pitch, yawrate, thrust)"; however, the "send_setpoint" method is not found in the class mentioned above. I was able to find this method "_set_vel_setpoint(0.0,_ 0.0, 2, 0) #(vx,vy,vz,yaw-rate)" but it does not allow me to control my desired parameters. If someone could point me in the right direction it would be highly appreciated.

krichardsson, 2023-04-14:
I did not find a good example code that shows the use of `send_setpoint()`, but you can see how it is called in some of the [examples](https://github.com/bitcraze/crazyflie-lib-python/tree/6370208d61cd959c1009971558b53adb43e3a24b/examples). In these examples it is called with `0, 0, 0, 0` to stop the motors.

AngelS18, 2023-04-21:
Thank you for your answer, actually I'm working on that.