bitcraze/discussions #1454: I can scan the crazyflie, but not connect to it

grappyschaefer, 2024-07-07:
### Discussed in https://github.com/orgs/bitcraze/discussions/1334

<div type='discussions-op-text'>

<sup>Originally posted by **socamo7005** May  9, 2024</sup>
I am trying to connect to the crazyflie via the virtual machine, I am able to scan the radio with the crazyradio PA, and able to click on connect. The M4 light starts flashing green, but the virtual machine says it is still connecting to the radio, it stays like that, and I get no data from the drone. I would appreciate the help.</div>

I think I just did something similar as I have a Crazyflie 2.1. While working through Droneblocks lessons, I used the Bolt instead of the cfc which said it failed as too many packets. I am using the CFClient on Mac Sonoma 14.5. I read how to reflash in Recovery mode but am not able to connect to the drone with the radio. The M2 and M3 are flashing blue. I have restarted the drone a few times with no change in behavior. I am at a loss on what to try next, so I would appreciate any ideas.
Thanks,
Karl

grappyschaefer, 2024-07-07:
Not sure if walking away helped but I just got it flashed and it is working. Great tutorial at https://www.bitcraze.io/documentation/repository/crazyflie-clients-python/master/userguides/recovery-mode/

closed 2024-07-07 by grappyschaefer

gemenerik, 2024-07-08:
Glad you got it working Karl!